stereolabs/zed-ros2-wrapper #258: Failed <<< zed_components [30.9s, exited with code 2]

luancvt, 2024-09-25:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

This seems to be an issue similar to [this one](https://github.com/stereolabs/zed-ros2-wrapper/issues/225).
I have followed the guide on the github page.
After running:

colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)

I receive:
Failed   <<< zed_components [30.9s, exited with code 2]
                                      
Summary: 1 package finished [1min 21s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed

With errors:
[image]

### Steps to Reproduce

1. Follow the guide
2. Fail at the colcon build part
3.
...

### Expected Result

Successfully installing the package

### Actual Result

[image]

### ZED Camera model

ZED

### Environment

```shell
Ubuntu 20.04
ARM
NVIDIA® Jetson™ Orin NX 16GB module
ZED SDK 4.1.4
ROS2 Foxy
```

### Anything else?

Can also be if I it was Zed SDK for Jetpack 5.1.1 or 5.1.2

Myzhar, 2024-09-25:
Pull and try to rebuild

closed 2024-09-25 by Myzhar

luancvt, 2024-09-26:
It worked! Thank you :+1:

commit 8f057ff8 (2024-09-25): Fix #258