stereolabs/zed-ros2-wrapper #264: ZED-Mini Longevity Test Bugs

annanya-tamu, 2024-10-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello,

I ran the zed-mini overnight with the zed ros2 wrapper as well. I also opened an instance of rviz2 so the Point Cloud topic and Image topics had subscribers. 

After a few hours, I saw this error come up multiple times:
`[zed_wrapper-2] [INFO] [1729118881.305305895] [zed.zed_node]: *** Odometry reset for LOOP CLOSURE event ***
` 
and eventually the wrapper threw this error:
```
[zed_wrapper-2] Sophus ensure failed in function 'static IOFTRACKER_Sophus::SO3<Scalar_> IOFTRACKER_Sophus::SO3<Scalar_, Options>::expAndTheta(const Tangent&, IOFTRACKER_Sophus::SO3<Scalar_, Options>::Scalar*) [with Scalar_ = float; int Options = 0; IOFTRACKER_Sophus::SO3<Scalar_, Options>::Tangent = Eigen::Matrix<float, 3, 1>; IOFTRACKER_Sophus::SO3<Scalar_, Options>::Scalar = float]', file '/builds/sl/ZEDKit/lib/include/sl_core/tracking/Tracker/IOFTracker/sophus/so3.hpp', line 612.
[ERROR] [zed_wrapper-2]: process has died [pid 154202, exit code -6, cmd '/home/ubuntu/Documents/ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed --params-file /home/ubuntu/Documents/rad_lab_marsupial_2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/ubuntu/Documents/ws/install/zed_wrapper/share/zed_wrapper/config/zedm.yaml --params-file /home/ubuntu/Documents/ws/install/zed_wrapper/share/zed_wrapper/config/ffmpeg.yaml --params-file /tmp/launch_params_6ht2ei8m'].
```
Somebody on another project suggested checking for NaN values in this function: https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/439#issuecomment-1096299467

Any help on this would be much appreciated!

### Steps to Reproduce

1. Connect zed mini to Orin nano
2. Run zed ros2 wrapper with default parameters
3. Run rviz2 with subscriptions to 1 Point Cloud and Left Image topic

### Expected Result

Both pointcloud and image topics to be running at the same frequency no matter how long wrapper has been up.

### Actual Result

Zed ros2 wrapper crashed with the error described above.

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 22.04
CPU: ARM
GPU: Nvidia Jetson Orin Nano
ZED SDK version: z4.2.1
ROS2 Distribution: Humble
```

Myzhar, 2024-10-18:
Are you using Positional Tracking `GEN_1` or `GEN_2`?

annanya-tamu, 2024-10-18:
`GEN_2`

Myzhar, 2024-10-18:
Please try if the problem persists with `GEN_1`

Myzhar, 2024-10-18:
`[zed_wrapper-2] [INFO] [1729118881.305305895] [zed.zed_node]: *** Odometry reset for LOOP CLOSURE event ***`
This is not an error. This informs you that a loop closure has been detected and the Odometry was reset to remove drift errors

closed 2024-11-24 by github-actions[bot]