stereolabs/zed-ros2-wrapper #265: ZED-Mini IMU Initialization

annanya-tamu, 2024-10-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello,

I ran the zed-mini for less than an hour with the zed ros2 wrapper as well. I also opened an instance of rviz2 so the Point Cloud topic and Image topics had subscribers.

The wrapper eventually threw this error:
```
[zed_wrapper-2] [2024-10-18 16:29:42 UTC][ZED][WARNING] IMU initialization failed, gyro bias too high. Found -0.05130885193141408 -0.042381061746106743 -0.04324537508178055 vs factory calibration 0.0009650529221373622 0.0006340666013872853 0.0019884070958669803
[ERROR] [zed_wrapper-2]: process has died [pid 224052, exit code -11, cmd '/home/ubuntu/Documents/ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed --params-file /home/ubuntu/Documents/ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/ubuntu/Documents/ws/install/zed_wrapper/share/zed_wrapper/config/zedm.yaml --params-file /home/ubuntu/Documents/rad_lab_marsupial_2_ws/install/zed_wrapper/share/zed_wrapper/config/ffmpeg.yaml --params-file /tmp/launch_params_vu94edf4']
```
Any help regarding this would be much appreciated!

### Steps to Reproduce

1. Connect zed mini to Orin nano
2. Run zed ros2 wrapper with default parameters
3. Run rviz2 with subscriptions to 1 Point Cloud and Left Image topic

### Expected Result

Both pointcloud and image topics to be running at the same frequency no matter how long wrapper has been up.

### Actual Result

Zed ros2 wrapper crashed with the error described above.

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 22.04
CPU: ARM
GPU: Nvidia Jetson Orin Nano
ZED SDK version: z4.2.1
ROS2 Distribution: Humble
```

### Anything else?

This is my `common.yaml` file:
```
# config/common_yaml
# Common parameters to Stereolabs ZED and ZED mini cameras

---
/**:
    ros__parameters:
        use_sim_time: false # Set to `true` only if there is a publisher for the simulated clock to the `/clock` topic. Normally used in simulation mode.

        simulation:
            sim_enabled: false # Set to `true` to enable the simulation mode and connect to a simulation server
            sim_address: "127.0.0.1" # The connection address of the simulation server. See the documentation of the supported simulation plugins for more information.
            sim_port: 30000 # The connection port of the simulation server. See the documentation of the supported simulation plugins for more information.

        svo:
            svo_loop: false # Enable loop mode when using an SVO as input source
            svo_realtime: true # if true the SVO will be played trying to respect the original framerate eventually skipping frames, otherwise every frame will be processed respecting the `pub_frame_rate` setting

        general:
            camera_timeout_sec: 5
            camera_max_reconnect: 5
[... 151 lines cut ...]
            debug_mapping: false
            debug_terrain_mapping: false
            debug_object_detection: false
            debug_body_tracking: false
            debug_roi: false
            debug_streaming: false
            debug_advanced: false
```
and this is my zedm.yaml file:
```
# config/zedm_yaml
# Parameters for Stereolabs ZED mini camera
---
/**:
    ros__parameters:
        general:
            camera_model: 'zedm'
            camera_name: 'zedm' # usually overwritten by launch file
            grab_resolution: 'HD720' # The native camera grab resolution. 'HD2K', 'HD1080', 'HD720', 'VGA', 'AUTO'
            grab_frame_rate: 15 # ZED SDK internal grabbing rate

        depth:
            min_depth: 0.1 # Min: 0.1, Max: 3.0
            max_depth: 5.0 # Max: 20.0
```

AhmedLHP, 2024-11-04:
I have the same 
Running ZED SDK : 4.2.1
OS: 20.04
Camera : 4 ZEDX with Quad Capture card
ROS: noetic

bsseo89, 2024-11-09:
Any solution yet?? 
Jetson Xavior Orin 64
Docker version 27.3.1, build ce12230

In side docker~~
Kernel version 5.15.136-tegra
# R36 (release), REVISION: 3.0, GCID: 36923193, BOARD: generic, EABI: aarch64, DATE: Fri Jul 19 23:24:25 UTC 2024
# KERNEL_VARIANT: oot
OpenCV Version: 4.5.4
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Aug_15_22:08:11_PDT_2023
Cuda compilation tools, release 12.2, V12.2.140
Build cuda_12.2.r12.2/compiler.33191640_0
ROS Humble 
Pytorch version : 2.1.0

closed 2025-01-18 by github-actions[bot]

k-jayanth, 2025-12-04:
i am facing same issue with zed x