stereolabs/zed-ros2-wrapper #266: problem installation

t-domenigoni, 2024-10-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

when from colcon build there is an error
why ?

### Steps to Reproduce

1.installation
2.
3.
...

### Expected Result

installation ok

### Actual Result

Starting >>> zed_components
--- stderr: zed_components                               
CMake Error at CMakeLists.txt:81 (find_package):
  By not providing "FindZED.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZED", but
  CMake did not find one.

  Could not find a package configuration file provided by "ZED" with any of
  the following names:

    ZEDConfig.cmake
    zed-config.cmake

  Add the installation prefix of "ZED" to CMAKE_PREFIX_PATH or set "ZED_DIR"
  to a directory containing one of the above files.  If "ZED" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< zed_components [1.08s, exited with code 1]

### ZED Camera model

ZED

### Environment

```shell
OS: Ubuntu 22.04
CPU: ARM
GPU: 
ZED SDK version: v4.2.1
ROS2 Distribution: Humble
```

### Anything else?

no

Myzhar, 2024-10-21:
Hi @t-domenigoni 
Did you install the ZED SDK correctly?
What's your host device? GPU Model?

t-domenigoni, 2024-10-21:
No, I have a problem when i run colcon build

t-domenigoni, 2024-10-21:
Host device : 
Cpux86_64
Gpu : intel hd graphics 500

Myzhar, 2024-10-21:
You need an NVIDIA GPU to use the ZED SDK and the ZED ROS2 Wrapper around it.

t-domenigoni, 2024-11-07:
> You need an NVIDIA GPU to use the ZED SDK and the ZED ROS2 Wrapper around it.

Hello i have a processor intel core i9 with nvidia geforce rtx4090. 
My laptop is dual boot with Windows And ubuntu. Maybe is it a problem?

Myzhar, 2024-11-07:
Hi @t-domenigoni 
please do not reply to closed issues. 
Furthermore, you are not reporting a bug, but you are posting a question. The right place for questions is the [Stereolabs forum](https://community.stereolabs.com/) or an email to support@stereolabs.com