stereolabs/zed-ros2-wrapper #267: Spatial Mapping Fused Point Cloud Not Populating

liamcarlos, 2024-10-22:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello there, I've trying to use the spatial mapping feature with my ZED One by making a service call to ~/enable_mapping but it doesn't seem to be working.

### Steps to Reproduce

1. Run rviz display launch file:
```
ros2 launch zed_display_rviz2 display_zed_cam.launch.py camera_model:=zed
```
2.  Make service call to enable mapping:
```
ros2 service call /zed/zed_node/enable_mapping std_srvs/srv/SetBool data:\ true\
```

### Expected Result

I expected to see the ~/fused_cloud pointcloud topic populate and show feedback in RViz

### Actual Result

Instead, nothing appears in RViz and when echoed the data field remains empty:

```
---
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: map
height: 1
width: 0
fields:
- name: x
  offset: 0
  datatype: 7
  count: 1
- name: y
  offset: 4
  datatype: 7
  count: 1
- name: z
  offset: 8
  datatype: 7
  count: 1
- name: rgb
  offset: 12
  datatype: 7
  count: 1
is_bigendian: false
point_step: 16
row_step: 0
data: []
is_dense: false
---
```

### ZED Camera model

ZED

### Environment

```shell
OS: Ubuntu 22.04 Jammy
CPU: x86_64
GPU: Nvidia GA106 [RTX A2000 12GB]
ZED SDK version: v4.2.1
Other info: ROS Iron
```

### Anything else?

The spatial_mapping.py sample that comes with the ZED SDK works fine.

closed 2024-11-28 by github-actions[bot]

neerja123-gupta, 2024-12-03:
Hi, I am having similar issues. Were you able to resolve it? Thanks

liamcarlos, 2024-12-03:
@neerja123-gupta  Yes, but I don't remember what I did..  Are you calling the spatial mapping service?

neerja123-gupta, 2024-12-04:
@liamcarlos , yes but its working now, it was the issue with cuda. Thanks for your reply.