stereolabs/zed-ros2-wrapper #268: Wrapper memory usage increasing over time

cholox, 2024-11-07:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello,

I'm trying to follow on the ros2 tutorials and have an error when running the zed_wrapper after some time. I searched for a solution but couldn't find one. After monitoring the memory usage with **jtop**, I noticed it is increasing over time. The wrapper dies when there is no memory left.

Here is the error:

```
1730999462.7335060 [DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0xffff9f73f2e0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0xffff9f73f130>'
1731001376.6479559 [ERROR] [zed_wrapper-2]: process has died [pid 115252, exit code -11, cmd '/home/ubuntu/ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed --params-file /home/ubuntu/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /home/ubuntu/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zed2i.yaml --params-file /home/ubuntu/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/ffmpeg.yaml --params-file /tmp/launch_params_pdwerm77'].
1731001376.6493154 [DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessExited'
1731001376.6500533 [DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0xffffa034f7c0>'
1731001376.6502960 [DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0xffffa034f7c0>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0xffff9f73eb00>'
1731001376.6505544 [DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0xffffa034f7c0>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0xffff9f73e4a0>'
```

There are no previous errors to this. 

I made a video of jtop:
[image]

### Steps to Reproduce

1. Install the wrapper with commit:
`469da81 (HEAD -> master, origin/master, origin/HEAD) Add missing 'object_detection.enable_tracking' to common.yaml`
2. Start zed_wrapper:
`ros2 launch -d zed_wrapper zed_camera.launch.py camera_model:=zed2i`

### Expected Result

The wrapper continues to run indefinitely.

### Actual Result

Memory usage increases until there is no memory left.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04
CPU: ARM
GPU: Nvidia Jetson Orin Nano
ZED SDK version: v4.2.1
ROS2 Distribution: Humble
```

### Anything else?

Here is my `common.yaml`:
```
# config/common_yaml
# Common parameters to Stereolabs ZED and ZED mini cameras

---
/**:
    ros__parameters:
        use_sim_time: false # Set to `true` only if there is a publisher for the simulated clock to the `/clock` topic. Normally used in simulation mode.

        simulation:
            sim_enabled: false # Set to `true` to enable the simulation mode and connect to a simulation server
[... 151 lines cut ...]
            thread_sched_policy: "SCHED_BATCH" # 'SCHED_OTHER', 'SCHED_BATCH', 'SCHED_FIFO', 'SCHED_RR' - NOTE: 'SCHED_FIFO' and 'SCHED_RR' require 'sudo'
            thread_grab_priority: 50 # ONLY with 'SCHED_FIFO' and 'SCHED_RR' - [1 (LOW) z-> 99 (HIGH)] - NOTE: 'sudo' required
            thread_sensor_priority: 70 # ONLY with 'SCHED_FIFO' and 'SCHED_RR' - [1 (LOW) z-> 99 (HIGH)] - NOTE: 'sudo' required
            thread_pointcloud_priority: 60 # ONLY with 'SCHED_FIFO' and 'SCHED_RR' - [1 (LOW) z-> 99 (HIGH)] - NOTE: 'sudo' required

        debug:
            sdk_verbose: 1 # Set the verbose level of the ZED SDK
            debug_common: false
            debug_sim: false
            debug_video_depth: false
            debug_camera_controls: false
            debug_point_cloud: false
            debug_positional_tracking: false
            debug_gnss: false
            debug_sensors: false
            debug_mapping: false
            debug_terrain_mapping: false
            debug_object_detection: false
            debug_body_tracking: false
            debug_roi: false
            debug_streaming: false
            debug_advanced: false
```

All other files are default as per installation.

cocobird231, 2024-11-12:
I've facing the same issue, and I've tried to change the `pos_tracking_mode` to `GEN_1` from `GEN_2` and the memory usage is stable now. Not the best solution, but it works for now.

Myzhar, 2024-11-12:
@cocobird231 @cholox are you using the latest ZED SDK v4.2.1?

cholox, 2024-11-12:
@Myzhar yes, SDK v4.2.1, installed on the same day I opened the issue

Myzhar, 2024-11-12:
OK, I reported the issue to the team. It will be addressed in the next patch release.

closed 2024-12-19 by github-actions[bot]