stereolabs/zed-ros2-wrapper #275: Missing docker image `dustynv/ros:humble-ros-base-l4t-r36.3.0`

beniaminopozzan, 2024-11-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When trying to build the docker image for Nvidia Jetson, L4T version 36.3.0 with ROS 2 Humble the build fails as the base image `dustynv/ros:humble-ros-base-l4t-r36.3.0` used in

https://github.com/stereolabs/zed-ros2-wrapper/blob/855b2494643cd9e6874867a5bfbebc3381764bf8/docker/Dockerfile.l4t-humble#L2

does not exists.

### Steps to Reproduce

1. clone the repo and `cd docker`
2. run `./jetson_build_dockerfile_from_sdk_and_l4T_version.sh l4t-r36.3.0 zedsdk4.2.2`

### Expected Result

The container image shall build without issues

### Actual Result

The build fails immediately as the base image does not exist.
```sh
Building dockerfile for l4t-r36.3.0 and ZED SDK zedsdk4.2.2
[+] Building 0.9s (2/2) FINISHED                                                                                                                                                                                                                                   docker:default
 => [internal] load build definition from Dockerfile.l4t-humble                                                                                                                                                                                                              0.1s
 => => transferring dockerfile: 4.46kB                                                                                                                                                                                                                                       0.0s
 => ERROR [internal] load metadata for docker.io/dustynv/ros:humble-ros-base-l4t-r36.3.0                                                                                                                                                                                     0.7s
------
 > [internal] load metadata for docker.io/dustynv/ros:humble-ros-base-l4t-r36.3.0:
------
Dockerfile.l4t-humble:4
--------------------
   2 |     ARG IMAGE_NAME=dustynv/ros:humble-ros-base-${L4T_VERSION}
   3 |     
   4 | >>> FROM ${IMAGE_NAME}
   5 |     
   6 |     ARG ZED_SDK_MAJOR=4
--------------------
ERROR: failed to solve: dustynv/ros:humble-ros-base-l4t-r36.3.0: failed to resolve source metadata for docker.io/dustynv/ros:humble-ros-base-l4t-r36.3.0: docker.io/dustynv/ros:humble-ros-base-l4t-r36.3.0: not found
```

### ZED Camera model

ZED

### Environment

```shell
Unrelated to the specific issue.
```

Myzhar, 2024-11-21:
Hi @beniaminopozzan 
Unfortunately, @dusty-nv has not yet released this version:
https://hub.docker.com/r/dustynv/ros/tags?name=humble-ros

dusty-nv, 2024-11-21:
Sorry @Myzhar @beniaminopozzan , I did not realize Stereolabs zed-ros2-wrapper dependent on this image.  I will rebuild and push it now.  I have still been pushing the ros-desktop images by default, but I suspect you want the minimal base image.

beniaminopozzan, 2024-11-21:
Hi @dusty-nv , thank you!
Yes please, the minimal base image would be perfect: `humble-ros-base-l4t-r36.3.0`
If I'm not asking too much, also `humble-ros-base-l4t-r36.4.0` would be quite useful!

dusty-nv, 2024-11-21:
OK, `dustynv/ros:humble-ros-base-l4t-r36.3.0` build and pushed 👍 

The R36.4 build of it it hit an error, I am trying again now...

closed 2024-12-28 by github-actions[bot]