stereolabs/zed-ros2-wrapper #279: Override QoS of image related topics

SM20598, 2024-11-22:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am trying to override QoS settings for the image topics. I saw that on October 1st there was a change in how the QoS is handled.
Running `ros2 param list` I get the the following QoS settings

```
qos_overrides./clicked_point.subscription.depth
qos_overrides./clicked_point.subscription.history
qos_overrides./clicked_point.subscription.reliability
qos_overrides./tf.publisher.depth
qos_overrides./tf.publisher.durability
qos_overrides./tf.publisher.history
qos_overrides./tf.publisher.reliability
qos_overrides./parameter_events.publisher.depth
qos_overrides./parameter_events.publisher.durability
qos_overrides./parameter_events.publisher.history
qos_overrides./parameter_events.publisher.reliability
qos_overrides./zed/plane.publisher.depth
qos_overrides./zed/plane.publisher.history
qos_overrides./zed/plane.publisher.reliability
qos_overrides./zed/plane_marker.publisher.depth
qos_overrides./zed/plane_marker.publisher.history
qos_overrides./zed/plane_marker.publisher.reliability
qos_overrides./zed/zed_node/atm_press.publisher.depth
qos_overrides./zed/zed_node/atm_press.publisher.history
qos_overrides./zed/zed_node/atm_press.publisher.reliability
qos_overrides./zed/zed_node/depth/depth_info.publisher.depth
qos_overrides./zed/zed_node/depth/depth_info.publisher.history
qos_overrides./zed/zed_node/depth/depth_info.publisher.reliability
qos_overrides./zed/zed_node/disparity/disparity_image.publisher.depth
qos_overrides./zed/zed_node/disparity/disparity_image.publisher.history
qos_overrides./zed/zed_node/disparity/disparity_image.publisher.reliability
qos_overrides./zed/zed_node/imu/data.publisher.depth
qos_overrides./zed/zed_node/imu/data.publisher.history
qos_overrides./zed/zed_node/imu/data.publisher.reliability
qos_overrides./zed/zed_node/imu/data_raw.publisher.depth
qos_overrides./zed/zed_node/imu/data_raw.publisher.history
qos_overrides./zed/zed_node/imu/data_raw.publisher.reliability
qos_overrides./zed/zed_node/imu/mag.publisher.depth
qos_overrides./zed/zed_node/imu/mag.publisher.history
qos_overrides./zed/zed_node/imu/mag.publisher.reliability
qos_overrides./zed/zed_node/left_cam_imu_transform.publisher.depth
qos_overrides./zed/zed_node/left_cam_imu_transform.publisher.history
qos_overrides./zed/zed_node/left_cam_imu_transform.publisher.reliability
qos_overrides./zed/zed_node/odom.publisher.depth
qos_overrides./zed/zed_node/odom.publisher.history
qos_overrides./zed/zed_node/odom.publisher.reliability
qos_overrides./zed/zed_node/path_map.publisher.depth
qos_overrides./zed/zed_node/path_map.publisher.history
qos_overrides./zed/zed_node/path_map.publisher.reliability
qos_overrides./zed/zed_node/path_odom.publisher.depth
qos_overrides./zed/zed_node/path_odom.publisher.history
qos_overrides./zed/zed_node/path_odom.publisher.reliability
qos_overrides./zed/zed_node/point_cloud/cloud_registered.publisher.depth
qos_overrides./zed/zed_node/point_cloud/cloud_registered.publisher.history
qos_overrides./zed/zed_node/point_cloud/cloud_registered.publisher.reliability
qos_overrides./zed/zed_node/point_cloud/cloud_registered/draco.publisher.depth
qos_overrides./zed/zed_node/point_cloud/cloud_registered/draco.publisher.history
qos_overrides./zed/zed_node/point_cloud/cloud_registered/draco.publisher.reliability
qos_overrides./zed/zed_node/point_cloud/cloud_registered/zlib.publisher.depth
qos_overrides./zed/zed_node/point_cloud/cloud_registered/zlib.publisher.history
qos_overrides./zed/zed_node/point_cloud/cloud_registered/zlib.publisher.reliability
qos_overrides./zed/zed_node/point_cloud/cloud_registered/zstd.publisher.depth
qos_overrides./zed/zed_node/point_cloud/cloud_registered/zstd.publisher.history
qos_overrides./zed/zed_node/point_cloud/cloud_registered/zstd.publisher.reliability
qos_overrides./zed/zed_node/pose.publisher.depth
qos_overrides./zed/zed_node/pose.publisher.history
qos_overrides./zed/zed_node/pose.publisher.reliability
qos_overrides./zed/zed_node/pose/status.publisher.depth
qos_overrides./zed/zed_node/pose/status.publisher.history
qos_overrides./zed/zed_node/pose/status.publisher.reliability
qos_overrides./zed/zed_node/pose_with_covariance.publisher.depth
qos_overrides./zed/zed_node/pose_with_covariance.publisher.history
qos_overrides./zed/zed_node/pose_with_covariance.publisher.reliability
qos_overrides./zed/zed_node/temperature/imu.publisher.depth
qos_overrides./zed/zed_node/temperature/imu.publisher.history
qos_overrides./zed/zed_node/temperature/imu.publisher.reliability
qos_overrides./zed/zed_node/temperature/left.publisher.depth
qos_overrides./zed/zed_node/temperature/left.publisher.history
qos_overrides./zed/zed_node/temperature/left.publisher.reliability
qos_overrides./zed/zed_node/temperature/right.publisher.depth
qos_overrides./zed/zed_node/temperature/right.publisher.history
qos_overrides./zed/zed_node/temperature/right.publisher.reliability
```

Following [the link in the release notes](https://design.ros2.org/articles/qos_configurability.html), I was able to change the QoS of the topic `/zed/zed_node/imu/data_raw` by reading a yaml file for `ros_param_override_path`, confirmed when checking the topics info in `--verbose`. But since in the above there is no image related params, I wasn't able to change their QoS.

### Steps to Reproduce

1. `ros2 launch zed_wrapper zed_camera.launch.py camera_model:="zed2"`
2. `ros2 param list`
...

### Expected Result

I expect to see options to override QoS of image related topics e.g. `left_raw/image_raw_color` and QoS to change when I load a yaml file.

### Actual Result

QoS of topics do not change, verified by calling `ros2 topic info /zed/zed_node/left_raw/image_raw_color/ --verbose`

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 22.04
System: Nvidia Jetson AGX Orin 64GB developer kit
ROS: Humble
ZED-Wrapper 4.2.1
ZED-SDK 4.2.1
```

Myzhar, 2024-11-22:
Please add your custom values for the following parameters in the `common_stereo.yaml` file:

```
qos_overrides./zed/zed_node/imu/data.publisher.depth
qos_overrides./zed/zed_node/imu/data.publisher.history
qos_overrides./zed/zed_node/imu/data.publisher.reliability
qos_overrides./zed/zed_node/imu/data_raw.publisher.depth
qos_overrides./zed/zed_node/imu/data_raw.publisher.history
qos_overrides./zed/zed_node/imu/data_raw.publisher.reliability
```

SM20598, 2024-11-22:
Hi @Myzhar,

I was able to override the QoS of the imu topics by changing the values in the `common_stereo.yaml`. But I can't change them for image related topics such as `left_raw/image_raw_color`:

```
# config/common_stereo.yaml
# Common parameters to Stereolabs ZED Stereo cameras

---
/**:
    ros__parameters:
        qos_overrides:
            '/zed/zed_node/zed_node/imu/data_raw':
                publisher:
                    reliability: best_effort
            '/zed/zed_node/left_raw/image_raw_color':
                publisher:
                    reliability: best_effort

        use_sim_time: false # Set to `true` only if there is a publisher for the simulated clock to the `/clock` topic. Normally used in simulation mode.

        simulation:
            sim_enabled: false # Set to `true` to enable the simulation mode and connect to a simulation server
            sim_address: "127.0.0.1" # The connection address of the simulation server. See the documentation of the supported simulation plugins for more information.
            sim_port: 30000 # The connection port of the simulation server. See the documentation of the supported simulation plugins for more information.

        svo:
            svo_loop: true # Enable loop mode when using an SVO as input source
            svo_realtime: false # if true the SVO will be played trying to respect the original framerate eventually skipping frames, otherwise every frame will be processed respecting the `pub_frame_rate` setting

        general:
            camera_timeout_sec: 5
            camera_max_reconnect: 5
            camera_flip: false
            self_calib: true # Enable the self-calibration process at camera opening. See https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#affeaa06cfc1d849e311e484ceb8edcc5
            serial_number: 0 # usually overwritten by launch file
            pub_resolution: "NATIVE" # The resolution used for output. 'NATIVE' to use the same `general.grab_resolution` - `CUSTOM` to apply the `general.pub_downscale_factor` downscale factory to reduce bandwidth in transmission
            pub_downscale_factor: 2.0 # rescale factor used to rescale image before publishing when 'pub_resolution' is 'CUSTOM'
            pub_frame_rate: 15.0 # frequency of publishing of visual images and depth images
            gpu_id: -1
            optional_opencv_calibration_file: "" # Optional path where the ZED SDK can find a file containing the calibration information of the camera computed by OpenCV. Read the ZED SDK documentation for more information: https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#a9eab2753374ef3baec1d31960859ba19

        video:
            brightness: 4 # [DYNAMIC] Not available for ZED X/ZED X Mini
            contrast: 4 # [DYNAMIC] Not available for ZED X/ZED X Mini
            hue: 0 # [DYNAMIC] Not available for ZED X/ZED X Mini
            saturation: 4 # [DYNAMIC]
            sharpness: 4 # [DYNAMIC]
            gamma: 8 # [DYNAMIC]
            auto_exposure_gain: true # [DYNAMIC]
            exposure: 80 # [DYNAMIC]
            gain: 80 # [DYNAMIC]
            auto_whitebalance: true # [DYNAMIC]
            whitebalance_temperature: 42 # [DYNAMIC] - [28,65] x100 - works only if `auto_whitebalance` is false

        sensors:
            publish_imu_tf: true # [usually overwritten by launch file] enable/disable the IMU TF broadcasting
            sensors_image_sync: false # Synchronize Sensors messages with latest published video/depth message
            sensors_pub_rate: 200. # frequency of publishing of sensors data. MAX: 400. - MIN: grab rate

        region_of_interest: # Only stereo cameras
            automatic_roi: false # Enable the automatic ROI generation to automatically detect part of the robot in the FoV and remove them from the processing. Note: if enabled the value of `manual_polygon` is ignored
            depth_far_threshold_meters: 2.5 # Filtering how far object in the ROI should be considered, this is useful for a vehicle for instance
            image_height_ratio_cutoff: 0.5 # By default consider only the lower half of the image, can be useful to filter out the sky
            #manual_polygon: '[]' # A polygon defining the ROI where the ZED SDK perform the processing ignoring the rest. Coordinates must be normalized to '1.0' to be resolution independent.
            #manual_polygon: "[[0.25,0.33],[0.75,0.33],[0.75,0.5],[0.5,0.75],[0.25,0.5]]" # A polygon defining the ROI where the ZED SDK perform the processing ignoring the rest. Coordinates must be normalized to '1.0' to be resolution independent.
            #manual_polygon: '[[0.25,0.25],[0.75,0.25],[0.75,0.75],[0.25,0.75]]' # A polygon defining the ROI where the ZED SDK perform the processing ignoring the rest. Coordinates must be normalized to '1.0' to be resolution independent.
            #manual_polygon: '[[0.5,0.25],[0.75,0.5],[0.5,0.75],[0.25,0.5]]' # A polygon defining the ROI where the ZED SDK perform the processing ignoring the rest. Coordinates must be normalized to '1.0' to be resolution independent.
            apply_to_depth: true # Apply ROI to depth processing
            apply_to_positional_tracking: true # Apply ROI to positional tracking processing
[... 150 lines cut ...]
```

But running `ros2 topic info /zed/zed_node/left_raw/image_raw_color --verbose` shows no changes to QoS reliability

```
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: zed_node
Node namespace: /zed/zed_node
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: 01.10.0d.39.4c.2f.c7.be.6e.bb.95.28.00.00.58.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (10)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 0
```

Myzhar, 2024-11-22:
Hi @SM20598 
this seems to be a limitation of the `image_transport` package that does not support this feature.
I'm searching for a workaround.
If you find a solution, please do not hesitate to report it.

Myzhar, 2024-11-22:
I've found that in Rolling the `image_transport::create_publisher` function supports ``, while in Humble it does not.
Rolling: https://github.com/ros-perception/image_common/blob/rolling/image_transport/src/image_transport.cpp#L63
Humble: https://github.com/ros-perception/image_common/blob/humble/image_transport/src/image_transport.cpp#L63

This will be supported when we upgrade to Jazzy:
https://github.com/ros-perception/image_common/blob/jazzy/image_transport/src/image_transport.cpp#L63

closed 2024-12-29 by github-actions[bot]