stereolabs/zed-ros2-wrapper #282: Invalid topic name error when launching zed_camera.launch.py

creix, 2024-12-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello, i have encountered this error when launching the wrapper with the command:
`ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i`

```
[component_container-2] [INFO] [1733414829.790032321] [zed.zed_node]: Camera-IMU Translation: 
[component_container-2]  -0.002 -0.023061 0.000217
[component_container-2] [INFO] [1733414829.790224267] [zed.zed_node]: Camera-IMU Rotation:
[component_container-2] FFFFD08A1670
[component_container-2] 0.999985 0.004895 -0.002594
[component_container-2] -0.004901 0.999986 -0.002006
[component_container-2] 0.002584 0.002019 0.999995
[component_container-2] 
[component_container-2] [INFO] [1733414829.790310928] [zed.zed_node]: *** Subscribers ***
[component_container-2] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidTopicNameError'
[component_container-2]   what():  Invalid topic name: topic name must not be empty string:
[component_container-2]   ''
[component_container-2]    ^
[component_container-2] 
[ERROR] [component_container-2]: process has died [pid 102219, exit code -6, cmd '/opt/ros/foxy/lib/rclcpp_components/component_container --ros-args -r __node:=zed_container -r __ns:=/zed'].
```

### Steps to Reproduce

1. `git clone  --recursive https://github.com/stereolabs/zed-ros2-wrapper.git`
2. `sudo apt update`
3. `rosdep update`
4. `rosdep install --from-paths src --ignore-src -r -y`
5. `colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)`
6. source the workspace

### Expected Result

The wrapper node is launched correctly

### Actual Result

This is the full log of after launching the node:

```
ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i
[INFO] [launch]: All log files can be found below /home/wrapper/.ros/log/2024-12-05-17-07-03-003395-wrapper-desktop-102203
[INFO] [launch]: Default logging verbosity is set to INFO
Using common configuration file: /home/wrapper/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common_stereo.yaml
[INFO] [robot_state_publisher-1]: process started with pid [102217]
[INFO] [component_container-2]: process started with pid [102219]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed_camera_center had 2 children
[robot_state_publisher-1] Link zed_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed_right_camera_optical_frame had 0 children
[robot_state_publisher-1] [INFO] [1733414824.703258184] [zed.zed_state_publisher]: got segment zed_camera_center
[robot_state_publisher-1] [INFO] [1733414824.703679550] [zed.zed_state_publisher]: got segment zed_camera_link
[robot_state_publisher-1] [INFO] [1733414824.703799717] [zed.zed_state_publisher]: got segment zed_left_camera_frame
[robot_state_publisher-1] [INFO] [1733414824.703904394] [zed.zed_state_publisher]: got segment zed_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1733414824.703965294] [zed.zed_state_publisher]: got segment zed_right_camera_frame
[robot_state_publisher-1] [INFO] [1733414824.704017776] [zed.zed_state_publisher]: got segment zed_right_camera_optical_frame
[component_container-2] [INFO] [1733414824.803875680] [zed.zed_container]: Load Library: /home/wrapper/ros2_ws/install/zed_components/lib/libzed_camera_component.so
[component_container-2] [INFO] [1733414825.055269395] [zed.zed_container]: Found class: rclcpp_components::NodeFactoryTemplate<stereolabs::ZedCamera>
[component_container-2] [INFO] [1733414825.055590948] [zed.zed_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<stereolabs::ZedCamera>
[component_container-2] [INFO] [1733414825.517992398] [zed.zed_node]: ********************************
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/zed/zed_node' in container '/zed/zed_container'
[component_container-2] [INFO] [1733414825.518255196] [zed.zed_node]:       ZED Camera Component 
[... 243 lines cut ...]
[component_container-2] 0.999985 0.004895 -0.002594
[component_container-2] -0.004901 0.999986 -0.002006
[component_container-2] 0.002584 0.002019 0.999995
[component_container-2] 
[component_container-2] [INFO] [1733414829.790310928] [zed.zed_node]: *** Subscribers ***
[component_container-2] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidTopicNameError'
[component_container-2]   what():  Invalid topic name: topic name must not be empty string:
[component_container-2]   ''
[component_container-2]    ^
[component_container-2] 
[ERROR] [component_container-2]: process has died [pid 102219, exit code -6, cmd '/opt/ros/foxy/lib/rclcpp_components/component_container --ros-args -r __node:=zed_container -r __ns:=/zed'].
```

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20 - L4T 35.4.1 - Jetpack 5.1.2
ZED SDK Version: v4.2.2
GPU: Nvidian Jetson Xavier
ROS: ROS2 Foxy
```

creix, 2024-12-09:
> Hi @creix the problem is caused by a problem with ROS 2 Foxy when loading parameters from the YAML files. I recommend you read[ this GitHub thread](https://github.com/ros2/rclcpp/issues/715) and modify the YAML files to solve the problem.

Hello, I read the thread and saw also your reply on the thread. What of the two approaches explained would you suggest to apply, the one with remapping or having plain YAML files? Will there be anything else that i have to modify in order to have everything functioning?

Myzhar, 2024-12-09:
Other users successfully solved the problem with "plain" YAML files.

creix, 2024-12-09:
I tried right now to remove all the nestings from the YAML files and rebuilding the package but i'm still having the same error. Do you have any other suggestion?

kamdkslansgka, 2024-12-11:
> I tried right now to remove all the nestings from the YAML files and rebuilding the package but i'm still having the same error. Do you have any other suggestion?我现在试图从YAML文件中删除所有嵌套并重建包，但我仍然有同样的错误。你还有其他建议吗？
Have you solved this problem? I also encountered this problem.

[component_container-2] [INFO] [1733895069.607470620] [zed.zed_node]: Camera-IMU Translation: 
[component_container-2]  -0.002 -0.023061 0.000217
[component_container-2] [INFO] [1733895069.607649215] [zed.zed_node]: Camera-IMU Rotation:
[component_container-2] 7FFF506D40F0
[component_container-2] 0.999985 0.005382 -0.000591
[component_container-2] -0.005386 0.999967 -0.006086
[component_container-2] 0.000558 0.006089 0.999981
[component_container-2] 
[component_container-2] [INFO] [1733895069.607899644] [zed.zed_node]: *** Subscribers ***
[component_container-2] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidTopicNameError'
[component_container-2]   what():  Invalid topic name: topic name must not be empty string:
[component_container-2]   ''
[component_container-2]    ^
[component_container-2] 
[ERROR] [component_container-2]: process has died [pid 140342, exit code -6, cmd '/opt/ros/foxy/lib/rclcpp_components/component_container --ros-args -r __node:=zed_container -r __ns:=/zed'].

creix, 2024-12-11:
Hello @kamdkslansgka , currently i was not able to solve this specific error. The workaround i've adopted is to use the docker container running Ubuntu 22 and ROS2 Humble.

ketill1, 2024-12-17:
@kamdkslansgka I had simular issue on humble. I solved it by updating the YAML files to the ones provided by example repo. Some missing parameter can cause the tobic name to become empty.

kamdkslansgka, 2024-12-17:
> Hello @kamdkslansgka , currently i was not able to solve this specific error. The workaround i've adopted is to use the docker container running Ubuntu 22 and ROS2 Humble.

OK, thank you for your reply

kamdkslansgka, 2024-12-17:
> @kamdkslansgka I had simular issue on humble. I solved it by updating the YAML files to the ones provided in the repo. Some missing parameter can make the topic name empty.

Thanks for your suggestion, I'll try it out.

Ad18756544864, 2024-12-18:
I also encountered the same problem

cuteredpwnda, 2025-01-09:
> Other users successfully solved the problem with "plain" YAML files.

Could you please give more context (or an example) to _which_ and _how_ YAML files have to be changed?
Because I did the same @creix did and still get this error.

youngeun-gadget, 2025-02-14:
> [@kamdkslansgka](https://github.com/kamdkslansgka) I had simular issue on humble. I solved it by updating the YAML files to the ones provided in the repo. Some missing parameter can make the topic name empty.

Where did you find it? Could you share it with me?

closed 2025-03-22 by github-actions[bot]

RonaldPavon, 2025-03-25:
Hello, im getting the same error on ROS2 Foxy on Ubuntu 20.04 on Jetson Orin AGX, did anyone find the proper solution to fix it?

Thanks in advance!