stereolabs/zed-ros2-wrapper #287: Frame corruption

mhubii, 2024-12-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I've been noticing a strange frame corruption recently, see below (topic: `zed_node/left/image_rect_color`):

[frame_corruption.webm](https://github.com/user-attachments/assets/90f7a7b1-fc3d-4995-aac4-937e035ac169)

Also see screenshots below

[image]

[image]

I haven't had these issues using the `zed2i` in the past, so this issue could be related to the ROS driver.

### Steps to Reproduce

```shell
ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i
```

### Expected Result

Frames

### Actual Result

Corrupted frames.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04
ZED SDK Version: 4.2.2
CUDA: 12.1
Release: humble-v4.1.4
```

mhubii, 2024-12-19:
closing this issue because it is quite difficult to pinpoint it to this driver.

gianlucapargaetzi, 2024-12-20:
No solution for this? I have the same issues

mhubii, 2025-01-04:
hi @gianlucapargaetzi, no, no solution. Happy to hear someone else is facing the issue. Re-opening it in the hopes there might be a fix. Again, it is hard to pinpoint this issue to the ZED camera, but I don't get this type of issue with the Realsense for example.

Myzhar, 2025-01-04:
Frame corruption is caused by the USB3 communication.
It does not happen with RealSense because it does not provide data at the same high resolution/frequency, so it uses less bandwidth.
Normally you can consider different causes for this type of issue.

Hardware:
- USB3 cable too long or bad cable extenders
- USB3 cable too close to motors or power cables
- USB3 hubs
- USB3 channel shared with other devices

Software:
- host device overloaded: the CPU cannot retrieve data from the USB3 buffer fast enough and it overflows causing data overwrite with consequent frame corruption

Can you recognize one of the conditions from this list?

mhubii, 2025-01-05:
Been using original hardware but can try to re-run the camera in isolation to test against CPU load hypothesis. Thank you for the hints and explanations.

tomasz-lewicki, 2025-01-09:
@Myzhar I have been seeing intermittent frame corruption issue with ZEDM as well on `Ubuntu 22.04` + `SDK 4.2.2` + `CUDA 12.2`/`CUDA 12.6`.

I've tried everything from the list above, I'm using the shortest ZED-branded USB cable I have. I unplugged all other USB devices. I've tried to pin it on CPU/GPU load. But I'm observing it on a pretty-high end system running only `/usr/local/bin/ZED_Explorer` at ~35% load on one CPU core.

[image]

[image]

closed 2025-02-15 by github-actions[bot]

mhubii, 2026-03-24:
the issue was closed due to staleness with no explanation provided. It might be beneficial for users to pin specific discussions on the forum