stereolabs/zed-ros2-wrapper #293: Unable to build ZED image using dustynv/ros:humble-ros-base-l4t-r36.3.0

leo-bonacini, 2025-01-13:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

While attempting to build the Docker image for NVIDIA Jetson Orin Nano (L4T version 36.3.0) with ROS 2 Humble, the build fails when using the base image `dustynv/ros:humble-ros-base-l4t-r36.3.0`, specified in the following Dockerfile line:

[Dockerfile Line 2](https://github.com/stereolabs/zed-ros2-wrapper/blob/855b2494643cd9e6874867a5bfbebc3381764bf8/docker/Dockerfile.l4t-humble#L2)

The error occurs at this line in the Dockerfile:
[Dockerfile Line 82](
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/docker/Dockerfile.l4t-humble#L82)

### Steps to Reproduce

1. `git clone https://github.com/stereolabs/zed-ros2-wrapper.git`
2. `cd /zed-ros2-wrapper/docker`
3. `./jetson_build_dockerfile_from_sdk_and_l4T_version.sh l4t-r36.3.0 zedsdk4.2.3`

### Expected Result

The image should build without any issues

### Actual Result

> > 309.9   apt: Failed to detect successful installation of [ros-humble-image-transport-plugins]
> 309.9   apt: Failed to detect successful installation of [ros-humble-compressed-image-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-compressed-depth-image-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-theora-image-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-ffmpeg-image-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-ffmpeg-encoder-decoder]
> 309.9   apt: Failed to detect successful installation of [ros-humble-point-cloud-transport-plugins]
> 309.9   apt: Failed to detect successful installation of [ros-humble-draco-point-cloud-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-zlib-point-cloud-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-zstd-point-cloud-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-zed-msgs]
> 309.9   apt: Failed to detect successful installation of [ros-humble-diagnostic-msgs]
> 309.9   apt: Failed to detect successful installation of [ros-humble-uncrustify-vendor]
> 309.9   apt: Failed to detect successful installation of [ros-humble-rclcpp-lifecycle]
> 309.9   apt: Failed to detect successful installation of [ros-humble-stereo-msgs]
> 309.9   apt: Failed to detect successful installation of [ros-humble-sensor-msgs]
> 309.9   apt: Failed to detect successful installation of [ros-humble-geometry-msgs]
> 309.9   apt: Failed to detect successful installation of [ros-humble-nav-msgs]
> 309.9   apt: Failed to detect successful installation of [ros-humble-tf2]
> 309.9   apt: Failed to detect successful installation of [ros-humble-tf2-ros]
> 309.9   apt: Failed to detect successful installation of [ros-humble-tf2-geometry-msgs]
> 309.9   apt: Failed to detect successful installation of [ros-humble-image-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-point-cloud-transport]
> 309.9   apt: Failed to detect successful installation of [ros-humble-std-srvs]
> 309.9   apt: Failed to detect successful installation of [ros-humble-visualization-msgs]
> 309.9   apt: Failed to detect successful installation of [clang-tidy]
> 309.9   apt: Failed to detect successful installation of [pyflakes3]
> 309.9   apt: Failed to detect successful installation of [ros-humble-launch-testing-ament-cmake]
> 309.9   apt: Failed to detect successful installation of [ros-humble-launch-testing-ros]
> 309.9   apt: Failed to detect successful installation of [ros-humble-pluginlib]
> 309.9   apt: Failed to detect successful installation of [ros-humble-rosidl-default-generators]
> 309.9   apt: Failed to detect successful installation of [libgeographic-dev]
> 309.9   apt: Failed to detect successful installation of [ros-humble-message-filters]
> 309.9   apt: Failed to detect successful installation of [ros-humble-tf2-eigen]
> 309.9   apt: Failed to detect successful installation of [ros-humble-yaml-cpp-vendor]

### ZED Camera model

ZED2i

### Environment

```shell
Unrelated to the specific issue
```

Myzhar, 2025-01-22:
@leo-bonacini can you please test the new scripts in the `update_docker` branch?

leo-bonacini, 2025-01-22:
**Report Procedure**

_System information:_
> NVIDIA NVIDIA Jetson Orin Nano Developer Kit
>  L4T 36.3.0 [ JetPack UNKNOWN ]
>    Ubuntu 22.04.4 LTS
>    Kernel Version: 5.15.136-tegra
>  CUDA 12.2.140
>    CUDA Architecture: 8.7
>  OpenCV version: 4.8.0
>    OpenCV Cuda: NO
>  CUDNN: 8.9.4.25
>  TensorRT: 8.6.2.3
>  Vision Works: NOT_INSTALLED
>  VPI: 3.1.5
>  Vulcan: 1.3.204

**Steps Followed**
1. Clone the zed-ros2-wrapper:
`git clone https://github.com/stereolabs/zed-ros2-wrapper.git`

2. Switch to the Required Branch:
`cd zed-ros2-wrapper && git checkout update_docker`

3. Build the Docker Image:
`./jetson_build_dockerfile_from_sdk_and_l4T_version.sh l4t-r36.3.0 zedsdk-4.2.3`

4. Run the Docker Container:
`docker run --runtime nvidia -it --privileged --network=host --ipc=host --pid=host   -e NVIDIA_DRIVER_CAPABILITIES=all -e DISPLAY=$DISPLAY   -v /tmp:/tmp   -v /dev:/dev   -v /var/nvidia/nvcam/settings/:/var/nvidia/nvcam/settings/   -v /etc/systemd/system/zed_x_daemon.service:/etc/systemd/system/zed_x_daemon.service   -v /usr/local/zed/resources/:/usr/local/zed/resources/   -v /usr/local/zed/settings/:/usr/local/zed/settings/   zed_ros2_l4t_36.3.0_sdk_4.2.3`

5. Run the ROS 2 Code:
`ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i`

Observations
- The image was successfully created, and the ROS 2 code runs as expected, even after installing and switching the middleware.
- ROS_DOMAIN_ID is set to 5 in _ros_entrypoint_jetson.sh_. The default value in ROS 2 is 0, I am just highlighting this detail might help other users.

Thank you for the update, @Myzhar. Let me know if further details are needed.

Myzhar, 2025-01-23:
Thank you for the feedback. We will merge this dev branch into the master branch as soon as we complete the QA tests.

We set ROS_DOMAIN_ID to 5 to be sure that this setup does not interfere with other robotics systems on the same local network.
I will update the README to highlight this behavior.

closed 2025-03-01 by github-actions[bot]