stereolabs/zed-ros2-wrapper #295: Save Area Map feature request

shawnhanna, 2025-01-16:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

I noticed that the zed-ros-wrapper has the ability to use the pos_tracking's zed->saveAreaMap function, but that hasn't been ported to ros2. I think this feature could greatly help adoption if it provides my robot the ability to localize in a global context.

### Use-Case

I would use it to localize my robot globally and use the map pose status information to know that I'm "locked on"

FlavioNzambi, 2025-05-26:
Was this added yet?

Myzhar, 2025-05-26:
> Was this added yet?

Not yet, we plan to add this feature very soon.

rohitdhote111, 2025-07-22:
is the service for saving 3d map from fused point cloud added?

Myzhar, 2025-07-22:
Yes it is.
Read the changelog:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/CHANGELOG.rst#2025-07-08

closed 2025-07-22 by Myzhar