stereolabs/zed-ros2-wrapper #307: Low framerate

shawnhanna, 2025-02-20:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am unable to get a decent framerate out of the camera when running the ROS2 driver. I am getting about 2-7hz output when I use `ros2 topic hz` on the image_rect_color topic for the camera

### Steps to Reproduce

1. ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedx
2. ros2 topic hz -w50 /zed/zed_node/left_raw/image_raw_color

### Expected Result

I have set the grab and pub rate to 15hz, so I expect ~15hz output of the images

### Actual Result

Getting ros2 topic hz results showing 2-7 hz on the left_raw/image_raw_color topic

### ZED Camera model

ZED

### Environment

```shell
OS: Jetson JP6.1
CPU: Jetson Orin AGX
GPU: Jetson Orin AGX
ZED SDK Version: v4.2.5
```

shawnhanna, 2025-02-20:
EDIT:

I see in the zed Explorer, I get frame drops too. I was misreading/looking. I get frame drops at every setting level, from HD1200 to SVGA, and all the FPS settings too.

For example, with HD1200 and 30FPS, I only get about 18 and see the frame drop increment a lot

Myzhar, 2025-02-20:
What resolution are you using?
What grab frame rate?
Are you using the latest version of the master branch?
What's the power mode of your Jetson?

Can you post the full node log?

Myzhar, 2025-02-20:
Did you follow our DDS tuning guide?
https://www.stereolabs.com/docs/ros2/150_dds_and_network_tuning

Please also use `rqt` to monitor the node statistic and check the node running frequency.

shawnhanna, 2025-02-20:
DDS tuning seems to fix it. Thanks! I thought I had already done that but it seems like the changes weren't persisting.

FYI: The /etc/sysctl.d/10-cyclone-max.conf settings don't seem to be applying in my docker container

I got around that by just manually call the commands when starting docker, so it's ok for now

closed 2025-03-30 by github-actions[bot]