stereolabs/zed-ros2-wrapper #308: Lifecycle Node Implementation

vijayreddysamula, 2025-02-21:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

LifecycleNode along with rclcpp::Node

### Use-Case

This feature helps alot for my robot to have stronger control over node states.

### Anything else?

Hi,
   I have concern regarding lifecycleNode setup. In the past, the zed has this support. But later removed it. It will be lot helpful. if the lifecycleNode is still available.

vijayreddysamula, 2025-02-25:
Hi @Myzhar .

Thank you for the update! I'm hoping that the development can move forward sooner. I'm looking forward to seeing the changes and improvements in the future releases.

Myzhar, 2026-03-02:
An update: 

- Lifecycle is not supported in Humble by `image_transport`, but it will be supported in newer versions.
- Lifecycle is not supported by NVIDIA Nitros: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nitros/issues/68
  - NVIDIA is working on it for future releases: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nitros/pull/70