stereolabs/zed-ros2-wrapper #310: Cannot Disable Position Tracking

tharp789, 2025-02-28:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am trying to limit the GPU usage taken up by disabling features using the common_stereo.yaml file using two ZED-X stereo cameras. 

It seems that in the YAML file, if I set pos_tracking_enabled: false, I am still able to see the pose of the camera in Rviz. It looks like the only way to fully disable the position tracking is to set the depth_mode: 'NONE' which causes the Pose to not be published. 

Is this a bug? or is there another way I should be disabling the position tracking? Unfortunately, I need depth but not position.  

### Steps to Reproduce

1. Set pos_tracking_enabled -> false
2. Launch zed_camera.launch.py 
3. Can still visualize and echo camera position topic from /zed_node/pose
...

### Expected Result

I expect no pose to be published when the position tracking is disabled. 

### Actual Result

The position tracking seems to still be active even when disabled. 

### ZED Camera model

ZED

### Environment

```shell
OS:l4t-36.4.0
Platform: Connecttech Jetson Orin AGX
ZED SDK: 4.2.5
ROS2 Wrapper Version: humble-v4.2.x
```

Myzhar, 2025-02-28:
To disable positional tracking you must also disable depth stabilization.
In this function you can see all the conditions that cause positional tracking to be automatically enabled:
https://github.com/stereolabs/zed-ros2-wrapper/blob/37a5542cb55619d8b00c6f1dc12a2df51c008549/zed_components/src/zed_camera/src/zed_camera_component.cpp#L9049

tharp789, 2025-02-28:
Thank you. I believe I should have disabled all the parameters specified in that file. However, it still seems that the pose is able to be visualized and mirror the cameras movements. Below are my config and launch files. If there is something I am missing please let me know.

[wire_common.zip](https://github.com/user-attachments/files/19033126/wire_common.zip)
[zed_camera.launch.zip](https://github.com/user-attachments/files/19033125/zed_camera.launch.zip)

Myzhar, 2025-03-02:
Can you start the node and send me the full log?

ba2sakal, 2025-03-11:
I had the same issue:

In the ZED Diagnostics it says inactive but if you subscribe the topic it continues publishing.
And in the diagnostics the odometry tracking becames "OK"

- key: Input mode
    value: Live Camera
  - key: Video/Depth
    value: 'Mean Frequency: 30.0 Hz (100.0%)'
  - key: Video/Depth
    value: 'Processing Time: 0.000715 sec (Max. 0.033 sec)'
  - key: Depth status
    value: ACTIVE
  - key: Depth mode
    value: NEURAL
  - key: Point Cloud
    value: Topic not subscribed
  - key: Manual ROI
    value: ENABLED
  - key: GNSS Fusion
    value: DISABLED
  - key: Odometry tracking status
    value: OK
  - key: Spatial Memory status
    value: 'OFF'
  - key: TF Odometry
    value: DISABLED
  - key: TF Pose
    value: DISABLED
  - key: Object Detection
    value: INACTIVE
  - key: Body Tracking
    value: INACTIVE
  - key: TF IMU
    value: 'Mean Frequency: 59.0 Hz'
  - key: IMU Sensor
    value: Topics not subscribed
  - key: Magnetometer Sensor
    value: N/A
  - key: Barometer Sensor
    value: N/A
  - key: Left CMOS Temp.
    value: N/A
  - key: Right CMOS Temp.
    value: N/A
  - key: Camera Temp.
    value: 27.6 °C
  - key: SVO Recording
    value: NOT ACTIVE
  - key: Streaming Server
    value: NOT ACTIVE
---
header:
  stamp:
    sec: 1741695873
    nanosec: 648995960
  frame_id: ''
status:
- level: "\0"
  name: 'zed_front: ZED Diagnostic'
  message: Camera grabbing
  hardware_id: 'Stereolabs camera: zed_front'
  values:
  - key: Uptime
    value: 0 days, 0 hours, 2 min, 7.2001 sec
  - key: Capture
    value: 'Mean Frequency: 29.9 Hz (99.8%)'
  - key: Capture
    value: 'Tot. Processing Time: 0.033307 sec (Max. 0.033 sec)'
  - key: Frame Drop rate
    value: 4/3806 (0.105097%)
  - key: Input mode
    value: Live Camera
  - key: Video/Depth
    value: 'Mean Frequency: 30.0 Hz (100.0%)'
  - key: Video/Depth
    value: 'Processing Time: 0.000700 sec (Max. 0.033 sec)'
  - key: Depth status
    value: ACTIVE
  - key: Depth mode
    value: NEURAL
  - key: Point Cloud
    value: Topic not subscribed
  - key: Manual ROI
    value: ENABLED
  - key: GNSS Fusion
    value: DISABLED
  - key: Odometry tracking status
    value: OK
  - key: Spatial Memory status
    value: 'OFF'
  - key: TF Odometry
    value: DISABLED
  - key: TF Pose
    value: DISABLED
  - key: Object Detection
    value: INACTIVE
  - key: Body Tracking
    value: INACTIVE
  - key: TF IMU
    value: 'Mean Frequency: 59.9 Hz'
  - key: IMU Sensor
    value: Topics not subscribed
  - key: Magnetometer Sensor
    value: N/A
  - key: Barometer Sensor
    value: N/A
  - key: Left CMOS Temp.
    value: N/A
  - key: Right CMOS Temp.
    value: N/A
  - key: Camera Temp.
    value: 27.6 °C
  - key: SVO Recording
    value: NOT ACTIVE
  - key: Streaming Server
    value: NOT ACTIVE

Myzhar, 2025-03-11:
Positional tracking is disabled in that case.
It is enabled automatically when you subscribe to the odom or pose topic.

ba2sakal, 2025-03-11:
But there is when I subscribe to the depth topic

```
key: Odometry tracking status
value: OK
```
the thing is with 1 camera downsampled 3, 30 fps, with neural depth it uses too much sources.

[image]

is it normal then?

here is my roslog 

`root@porsche-desktop:/workspaces/isaac_ros-dev# ./camera.sh 
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-03-11-14-23-19-991703-porsche-desktop-297696
[INFO] [launch]: Default logging verbosity is set to INFO
launch_ros.description.ParameterValue(value=[<launch.substitutions.command.Command object at 0xffffbb4be020>], value_type=<class 'str'>)
[INFO] [launch.user]: * Starting Composable node container: /zed_multi/isaac_ros
[INFO] [launch.user]: * Starting ZED ROS2 node for camera zed_front (zedx/41197812)
[INFO] [launch.user]: Using common configuration file: /workspaces/isaac_ros-dev/install/zed_wrapper/share/zed_wrapper/config/common_stereo.yaml
[INFO] [launch.user]: Using camera configuration file: /workspaces/isaac_ros-dev/install/zed_wrapper/share/zed_wrapper/config/zedx.yaml
[INFO] [launch.user]: Using FFMPEG configuration file: /workspaces/isaac_ros-dev/install/zed_wrapper/share/zed_wrapper/config/ffmpeg.yaml
[INFO] [launch.user]: Loading ZED node `zed_front` in container `/zed_multi/isaac_ros`
[INFO] [component_container_mt-1]: process started with pid [297736]
[INFO] [robot_state_publisher-2]: process started with pid [297738]
[INFO] [robot_state_publisher-3]: process started with pid [297740]
[INFO] [joint_state_publisher-4]: process started with pid [297742]
[component_container_mt-1] [INFO] [1741699400.578740153] [zed_multi.isaac_ros]: Load Library: /workspaces/isaac_ros-dev/install/zed_components/lib/libzed_camera_component.so
[robot_state_publisher-3] [INFO] [1741699400.588761350] [zed_multi_state_publisher]: got segment C01
[robot_state_publisher-3] [INFO] [1741699400.588915556] [zed_multi_state_publisher]: got segment C02
[robot_state_publisher-3] [INFO] [1741699400.588935011] [zed_multi_state_publisher]: got segment C03
[robot_state_publisher-3] [INFO] [1741699400.588944995] [zed_multi_state_publisher]: got segment C04
[robot_state_publisher-3] [INFO] [1741699400.588952419] [zed_multi_state_publisher]: got segment C05
[robot_state_publisher-3] [INFO] [1741699400.588959555] [zed_multi_state_publisher]: got segment C06
[robot_state_publisher-3] [INFO] [1741699400.588966723] [zed_multi_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1741699400.588974115] [zed_multi_state_publisher]: got segment chassis_link
[robot_state_publisher-3] [INFO] [1741699400.588981635] [zed_multi_state_publisher]: got segment front_b_mount
[robot_state_publisher-3] [INFO] [1741699400.588988610] [zed_multi_state_publisher]: got segment front_c_mount
[robot_state_publisher-3] [INFO] [1741699400.588995522] [zed_multi_state_publisher]: got segment front_left_wheel_link
[robot_state_publisher-3] [INFO] [1741699400.589002978] [zed_multi_state_publisher]: got segment front_mount
[robot_state_publisher-3] [INFO] [1741699400.589010082] [zed_multi_state_publisher]: got segment front_right_wheel_link
[robot_state_publisher-3] [INFO] [1741699400.589017410] [zed_multi_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1741699400.589024770] [zed_multi_state_publisher]: got segment mid_mount
[robot_state_publisher-3] [INFO] [1741699400.589031970] [zed_multi_state_publisher]: got segment rear_b_mount
[robot_state_publisher-3] [INFO] [1741699400.589039202] [zed_multi_state_publisher]: got segment rear_c_mount
[robot_state_publisher-3] [INFO] [1741699400.589046369] [zed_multi_state_publisher]: got segment rear_left_wheel_link
[robot_state_publisher-3] [INFO] [1741699400.589053345] [zed_multi_state_publisher]: got segment rear_mount
[robot_state_publisher-3] [INFO] [1741699400.589060353] [zed_multi_state_publisher]: got segment rear_right_wheel_link
[robot_state_publisher-3] [INFO] [1741699400.589067201] [zed_multi_state_publisher]: got segment riser
[robot_state_publisher-3] [INFO] [1741699400.589074017] [zed_multi_state_publisher]: got segment riser_base_link
[robot_state_publisher-3] [INFO] [1741699400.589102528] [zed_multi_state_publisher]: got segment zed_back_camera_link
[robot_state_publisher-3] [INFO] [1741699400.589112576] [zed_multi_state_publisher]: got segment zed_front_camera_link
[robot_state_publisher-3] [INFO] [1741699400.589120480] [zed_multi_state_publisher]: got segment zed_left_camera_link
[robot_state_publisher-3] [INFO] [1741699400.589127392] [zed_multi_state_publisher]: got segment zed_multi_link
[robot_state_publisher-3] [INFO] [1741699400.589135616] [zed_multi_state_publisher]: got segment zed_right_camera_link
[robot_state_publisher-2] [INFO] [1741699400.592358406] [zed_multi.zed_front_state_publisher]: got segment zed_front_camera_center
[robot_state_publisher-2] [INFO] [1741699400.592492356] [zed_multi.zed_front_state_publisher]: got segment zed_front_camera_link
[robot_state_publisher-2] [INFO] [1741699400.592511940] [zed_multi.zed_front_state_publisher]: got segment zed_front_left_camera_frame
[robot_state_publisher-2] [INFO] [1741699400.592522692] [zed_multi.zed_front_state_publisher]: got segment zed_front_left_camera_optical_frame
[robot_state_publisher-2] [INFO] [1741699400.592530659] [zed_multi.zed_front_state_publisher]: got segment zed_front_right_camera_frame
[... cut ...]
[component_container_mt-1] [INFO] [1741699405.733270125] [zed_multi.zed_front]: Camera-IMU Rotation:
[component_container_mt-1] FFFF91FBC6D0
[component_container_mt-1] 0.999991 -0.004148 -0.000464
[component_container_mt-1] 0.004147 0.999984 -0.003761
[component_container_mt-1] 0.000479 0.003759 0.999993
[component_container_mt-1] 
[component_container_mt-1] [INFO] [1741699405.733296684] [zed_multi.zed_front]: *** Subscribers ***
[component_container_mt-1] [INFO] [1741699405.734211575] [zed_multi.zed_front]:  * Plane detection: '/clicked_point'
`

Myzhar, 2025-03-11:
I confirm that the positional tracking is not enabled.
If it was enabled, at the end of this log you would have found the information concerning the camera pose initialization

Myzhar, 2025-03-11:
Yes, NEURAL depth mode uses AI and this GPU usage is expected

ba2sakal, 2025-03-11:
Got it thanks!

closed 2025-03-24 by tharp789