stereolabs/zed-ros2-wrapper #316: ROS2 Build Error:

andrewjong, 2025-03-20:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm trying to colcon build the zed-ros2-wrapper on my desktop. I'm trying to build it within the `nvidia/cuda:12.6.3-devel-ubuntu22.04` docker image.

This error occurs:
```
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/zed_camera_one_component.dir/build.make:360: libzed_camera_one_component.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:192: CMakeFiles/zed_camera_one_component.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:360: libzed_camera_component.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:166: CMakeFiles/zed_camera_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_components [43.8s, exited with code 2]
```

Any idea how I can overcome this?

### Steps to Reproduce

1. Run `nvidia/cuda:12.6.3-devel-ubuntu22.04` docker image
2. Install dependencies for zed-ros2-wrapper
3. Try and colcon build

### Expected Result

It to build

### Actual Result

It didn't build

### ZED Camera model

ZED

### Environment

```shell
Docker image: nvidia/cuda:12.6.3-devel-ubuntu22.04

Host: Ubuntu 22.04
```

Myzhar, 2025-03-20:
What version of the ZED SDK are you downloading inside the container?
Can you share your dockerfile?

andrewjong, 2025-03-20:
This is the relevant part of my Dockerfile:
```Dockerfile
WORKDIR /tmp/

RUN apt update && apt install -y ros-humble-zed-msgs ros-humble-cob-srvs && rm -rf /var/lib/apt/lists/*

RUN if [ "$REAL_ROBOT" = "true" ]; then \
    wget -O zed_sdk.zstd.run https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/4.2/ZED_SDK_Ubuntu22_cuda12.1_v4.2.5.zstd.run ;\
  else \
    wget -O zed_sdk.zstd.run https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/4.2/ZED_SDK_Tegra_L4T36.4_v4.2.5.zstd.run ;\
  fi

RUN chmod +x zed_sdk.zstd.run && \
  ./zed_sdk.zstd.run -- silent runtime_only skip_od_module skip_cuda && rm zed_sdk.zstd.run

```

I just noticed the ZED_SDK download matches to CUDA 12.1. Do you think I need to switch my NVIDIA CUDA version to CUDA 12.1 instead of 12.6?

andrewjong, 2025-03-20:
I'm an idiot, I was downloading the wrong one by mistake in my dockerfile. Thanks!

closed 2025-03-20 by andrewjong