stereolabs/zed-ros2-wrapper #322: Image Flipping - Camera  ZED mini

andre-rebelo-teixeira, 2025-04-03:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I was running the ros2_wrapper node to get images from my camera, launched with the following command.
```bash 
```
As using foxglove as a platform to see the topics. I came across the issue of the image flipping sometimes at random, and the point cloud for that frame would be completly random, a video in slow mo is attached, the error can be seen in second ~11

https://github.com/user-attachments/assets/25495793-317f-4f7d-a28e-b26d2d369c5a

### Steps to Reproduce

1. ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedm publish_pointcloud:=true confidence:=99 
2. ros2 run foxglove_bridge  foxglove_bridge
3. Add Image topic to fox glove
...

### Expected Result

Image Stays in the same orientation always

### Actual Result

The results of the flipping can be seen in the video

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 22.04.5 LTS - 6.8.0-51-generic
CPU: 13th Gen Intel(R) Core(TM) i7-13620H
GPU: NVIDIA GeForce RTX 4060 Laptop GPU
ZED SDK: 5.0.0
CUDA Version: V12.2.140
CUDA Version: V12.6.85
ROS_DISTRO: ROS2 humble
```

Myzhar, 2025-04-03:
That's a random behavior happening when the USB3 buffer is saturated or the USB3 communication is not stable.
Are you using the original cable?

andre-rebelo-teixeira, 2025-04-03:
@Myzhar as a way of making sure my USB bandwidth was enough i ran the 'ZED_Diagnostic' Tool from the ZED SDK, and all the results pass including ``` Devices ``` -> ``` USB Bandwidth ```.

The full .json is affixed bellow in case it may be useful 

[ZED_Diagnostic_Results.json](https://github.com/user-attachments/files/19588795/ZED_Diagnostic_Results.json)

closed 2025-04-03 by andre-rebelo-teixeira