stereolabs/zed-ros2-wrapper #325: Cannot build on ROS2 Jazzy Jalisco

YuhsiHu, 2025-04-13:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I tried to build on ROS2 Jazzy Jalisco (Ubuntu 24) but it failed. Building `zed_msgs` from source is tried and did not change the result.

### Steps to Reproduce

1.install ROS2 and I am sure it works 
2. ZED2 camera SDK also works
3. Follow the README:
```bash
mkdir -p ~/ros2_ws/src/ # create your workspace if it does not exist
cd ~/ros2_ws/src/ #use your current ros2 workspace folder
git clone https://github.com/stereolabs/zed-ros2-wrapper.git
cd ..
sudo apt update
rosdep update
rosdep install --from-paths src --ignore-src -r -y # install dependencies
```
4. Try to build via:
```bash
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc) 
```
...

### Expected Result

Finish building.

### Actual Result

Failed.
```
Starting >>> zed_components
--- stderr: zed_components                               
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:43:2: error: #error Unsupported ROS2 distro
   43 | #error Unsupported ROS2 distro
      |  ^~~~~
In file included from /opt/ros/jazzy/include/rcl/rcl/timer.h:34,
                 from /opt/ros/jazzy/include/rcl/rcl/wait.h:33,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/context.hpp:33,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/contexts/default_context.hpp:18,
                 from /opt/ros/jazzy/include/rclcpp/rclcpp/clock.hpp:22,
                 from /home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/sl_tools.hpp:20,
                 from /home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:23,
                 from /home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:15:
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::startCamera()’:
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:4249:21: warning: format ‘%d’ expects argument of type ‘int’, but argument 5 has type ‘size_t’ {aka ‘long unsigned int’} [-Wformat=]
 4249 |       get_logger(), " * SVO resolution -> %dx%d",
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
 4250 |       mZed->getCameraInformation().camera_configuration.resolution.width,
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                    |
      |                                                                    size_t {aka long unsigned int}
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:4249:44: note: format string is defined here
 4249 |       get_logger(), " * SVO resolution -> %dx%d",
      |                                           ~^
      |                                            |
      |                                            int
      |                                           %ld
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:4249:21: warning: format ‘%d’ expects argument of type ‘int’, but argument 6 has type ‘size_t’ {aka ‘long unsigned int’} [-Wformat=]
 4249 |       get_logger(), " * SVO resolution -> %dx%d",
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
 4250 |       mZed->getCameraInformation().camera_configuration.resolution.width,
 4251 |       mZed->getCameraInformation().camera_configuration.resolution.height);
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                    |
      |                                                                    size_t {aka long unsigned int}
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:4249:47: note: format string is defined here
 4249 |       get_logger(), " * SVO resolution -> %dx%d",
      |                                              ~^
      |                                               |
      |                                               int
      |                                              %ld
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::processGeoPose()’:
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:8234:18: error: no matching function for call to ‘toMsg(tf2::Transform&)’
 8234 |       (tf2::toMsg(mMap2UtmTransf)) :
      |        ~~~~~~~~~~^~~~~~~~~~~~~~~~
In file included from /opt/ros/jazzy/include/tf2_ros/tf2_ros/buffer_interface.h:43,
                 from /opt/ros/jazzy/include/tf2_ros/tf2_ros/buffer.h:43,
                 from /home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/sl_types.hpp:18,
                 from /home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:24:
/opt/ros/jazzy/include/tf2/tf2/convert.hpp:129:3: note: candidate: ‘template<class A, class B> B tf2::toMsg(const A&)’
  129 | B toMsg(const A & a);
      |   ^~~~~
/opt/ros/jazzy/include/tf2/tf2/convert.hpp:129:3: note:   template argument deduction/substitution failed:
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:8234:18: note:   couldn’t deduce template parameter ‘B’
 8234 |       (tf2::toMsg(mMap2UtmTransf)) :
      |        ~~~~~~~~~~^~~~~~~~~~~~~~~~
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:8235:18: error: no matching function for call to ‘toMsg(tf2::Transform)’
 8235 |       (tf2::toMsg(mMap2UtmTransf.inverse()));
      |        ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/jazzy/include/tf2/tf2/convert.hpp:129:3: note: candidate: ‘template<class A, class B> B tf2::toMsg(const A&)’
  129 | B toMsg(const A & a);
      |   ^~~~~
/opt/ros/jazzy/include/tf2/tf2/convert.hpp:129:3: note:   template argument deduction/substitution failed:
/home/yuxihu/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:8235:18: note:   couldn’t deduce template parameter ‘B’
 8235 |       (tf2::toMsg(mMap2UtmTransf.inverse()));
      |        ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:118: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:166: CMakeFiles/zed_camera_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_components [15.7s, exited with code 2]

Summary: 0 packages finished [15.8s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed

```

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 24.04.2 LTS
CPU: Intel® Core™ i9-14900HX × 32
GPU: RTX 4080
ROS: ROS2 Jazzy Jalisco
```

YuhsiHu, 2025-04-13:
I think the code `zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp` need to be updated to support ROS2 Jazzy Jalisco:
```cpp
#elif defined FOUND_JAZZY
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
```
then it should be:
```cpp
#ifdef FOUND_HUMBLE
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
#elif defined FOUND_JAZZY
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
#elif defined FOUND_IRON
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
#elif defined FOUND_FOXY
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
#else
#error Unsupported ROS2 distro
```

YuhsiHu, 2025-04-13:
> [@YuhsiHu](https://github.com/YuhsiHu) what branch? Master or Jazzy?

the master branch.

Myzhar, 2025-04-13:
Add this line

```
#elif defined FOUND_JAZZY
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
```
here
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_components/src/zed_camera/src/zed_camera_component.cpp#L40

I will fix it on Monday

YuhsiHu, 2025-04-14:
Thank you for your quick [response](https://github.com/stereolabs/zed-ros2-wrapper/commit/8b9d14c9020d13013144a2f029b38e5e614d8f45)!

closed 2025-04-14 by YuhsiHu