stereolabs/zed-ros2-wrapper #340: Colcon build failed on both x86 and jetson orin NX

aravindsairam, 2025-04-30:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Im trying to build the ZED ros2 package in both x86 and jetson Orin NX 16 GB, but I get the same error in both system.

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

Build should be completed without any errors

### Actual Result

`Starting >>> zed_components
[Processing: zed_components]                             
--- stderr: zed_components                               
/workspaces/isaac_ros-dev/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::publishHealthStatus()’:
/workspaces/isaac_ros-dev/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:11696:35: error: ‘struct sl::HealthStatus’ has no member named ‘low_image_quality’
11696 |   msg->low_image_quality = status.low_image_quality;
      |                                   ^~~~~~~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:11697:30: error: ‘struct sl::HealthStatus’ has no member named ‘low_lighting’
11697 |   msg->low_lighting = status.low_lighting;
      |                              ^~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:11698:39: error: ‘struct sl::HealthStatus’ has no member named ‘low_depth_reliability’
11698 |   msg->low_depth_reliability = status.low_depth_reliability;
      |                                       ^~~~~~~~~~~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:11700:12: error: ‘struct sl::HealthStatus’ has no member named ‘low_motion_sensors_reliability’
11700 |     status.low_motion_sensors_reliability;
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:118: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:166: CMakeFiles/zed_camera_component.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_components [41.1s, exited with code 2]

Summary: 0 packages finished [42.0s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  1 package not processed
`

### ZED Camera model

ZED

### Environment

```shell
Ubuntu 22.04 and in Jetpack 6.2
ROS humble
Cuda 12.6
Isaac ROS workspace 3.2
```

Myzhar, 2025-04-30:
What ZED SDK version are you using?
Are you using the latest update in the master branch?

aravindsairam, 2025-04-30:
Yes, Im using the lastest update from from the master branch 
`git clone --recurse-submodules https://github.com/stereolabs/zed-ros2-wrapper`

ZED SDK version : 4.2