stereolabs/zed-ros2-wrapper #341: ROS2 Foxy build issue

Harin99, 2025-05-02:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I attempted to build the zed-ros2-wrapper in order to use the ZED camera in a ROS2 foxy. 
However, I encountered the following issues during the build process. 

Below are the steps I took in my setup. 

### Steps to Reproduce

1. NVIDIA Jetson ZED_SDK 4.0 Installation (https://www.stereolabs.com/en-kr/developers/release/4.0)
> ZED SDK for JetPack 5.1.2 (L4T 35.4)

2. Follow below steps: 
mkdir -p ~/ros2_ws/src 
cd ~/ros2_ws/src
git clone https://github.com/stereolabs/zed-ros2-wrapper.git 
cd zed-ros2-wrapper 
git checkout foxy-humble-v4.0.5 # To install the ZED SDK that is compatible with CUDA 11.4
cd ../
git clone https://github.com/ros-drivers/nmea_msgs.git
cd nmea_msgs 
git checkout 2.0.0 # for foxy 
cd ../
rosdep install --from-paths src --ignore-src -r -y
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)
3. After performing "colcon build~", i encountered the below error. 

> ac@AC:/ros2_ws$ colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release --parallel-workers $(nproc)
Starting >>> nmea_msgs
Finished <<< nmea_msgs [1.02s]                     
Starting >>> zed_components
--- stderr: zed_components                               
In file included from /home/ac/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:19,
                 from /home/ac/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:20:
/usr/local/zed/include/sl/Fusion.hpp:68:64: note: #pragma message: ~ FUSION SDK is distributed in Early Access ~
   68 | #pragma message("~ FUSION SDK is distributed in Early Access ~")
      |                                                                ^
/home/ac/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::startPosTracking()’:
/home/ac/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:4042:12: error: ‘struct sl::PositionalTrackingFusionParameters’ has no member named ‘gnss_initialisation_distance’
 4042 |     params.gnss_initialisation_distance = mGnssInitDistance;
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ac/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::processGeoPose()’:
/home/ac/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:6546:38: error: cannot convert ‘sl::GNSS_CALIBRATION_STATE’ to ‘sl::POSITIONAL_TRACKING_STATE’ in assignment
 6546 |   mGeoPoseStatus = mFusion.getGeoPose(mLastGeoPose);
      |                    ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
      |                                      |
      |                                      sl::GNSS_CALIBRATION_STATE
make[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:89: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/zed_camera_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Failed   <<< zed_components [19.2s, exited with code 2]
Summary: 1 package finished [20.7s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed
...

### Expected Result

.

### Actual Result

.

### ZED Camera model

ZED

### Environment

```shell
- Jetson Orin AGX (ARM)
- Ubuntu 20.04 + ROS2 foxy 
- CUDA 11.4 
- ZED SDK v4.0
```

### Anything else?

I suspect there is a version mismatch between the ZED_SDK and the zed-ros2-wrapper.
To verify this, I tried installing and testing multiple versions of the SDK — 4.0, 4.1, and 4.2 — but in all cases, the gnss_initialisation_distance field was not declared within Fusion.hpp, resulting in the same build error.
(Additionally, I encountered other issues, such as a type mismatch involving sl::GNSS_CALIBRATION_STATE.)

Is there any recommended solution for this issue, or could there be something wrong with the steps I followed?

Myzhar, 2025-05-06:
Hi @Harin99 
If you are not using GNSS fusion, you can comment out that part.
If you are using GNSS fusion, I recommend you do not use this old version of the Wrapper

Harin99, 2025-05-06:
@Myzhar Thank you so much for your support! I have two follow-up questions:

1. I am currently using the `foxy-humble-v4.0.5` tag to support ROS2 Foxy(with CUDA 11.4 & Jetpack 5.1.2). If I want to enable **GNSS fusion** in ROS2 Foxy, which **wrapper tag/branch** and **ZED SDK version** should I use for proper compatibility?
2. If I do **not** use GNSS fusion with the current setup (`foxy-humble-v4.0.5` wrapper and ZED SDK v4.0), would it be sufficient to **comment out all GNSS-related functions** defined within the `ZedCamera` class in the `zed_camera_component.cpp` file to resolve build issues?

Harin99, 2025-05-07:
@Myzhar Thank you for your reply! 

I have built the latest `zed-ros2-wrapper` and `zed-ros2-interfaces` packages using the most recent versions. Although I encountered some conflicts during the build, I was able to resolve them by commenting out the relevant parts, and the build eventually succeeded.

However, when I run the following command:

```bash
ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zedx
```

I encounter the same error described in this GitHub issue: [[#341](https://github.com/stereolabs/zed-ros2-wrapper/issues/341)](https://github.com/stereolabs/zed-ros2-wrapper/issues/341).
(i.e., what():  Invalid topic name: topic name must not be empty string:) 

I’ve read through the thread, but I’m still unclear about *which* `.yaml` file needs to be modified and *how exactly* it should be changed.

Could someone kindly explain this in more detail? I would really appreciate any guidance. Thank you in advance!

Harin99, 2025-05-07:
I successfully launched the ROS2 node and was able to receive image data via ROS2 topics using the following setup:
1. ROS2 wrapper: `foxy-humble-v4.0.5`
2. Interfaces: `foxy-humble-v4.0.5`
3. ZED SDK: `v4.0`

Although the GNSS-related error mentioned earlier still occurred, I was able to bypass it by commenting out the relevant code, as I do not use GNSS Fusion in my application.

Thank you for your support!

odd-incubus, 2025-05-15:
Hi, @Harin99 I am facing the same issue, can you share all the parts you commented? I tried doing it but it started causing more errors. 
Thanks in advance!

closed 2025-06-21 by github-actions[bot]