stereolabs/zed-ros2-wrapper #353: Random "Reset" of Data Streams

alec-jensen, 2025-06-07:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Everything sort of "resets" randomly, depth, point cloud, rgb, odometry, object tracking. it randomly jumps for one frame before returning to normal, throwing weird odometry values causing my app using the zed sdk to have errors.

### Steps to Reproduce

1. Launch the wrapper with this launch file: https://github.com/MHSeals/mhsboat_ctrl/blob/main/launch/camera_rviz.launch.py
2. Wait and see the issues

### Expected Result

Everything should run smoothly without these random interuptions.

### Actual Result

There are random interruptions of all data sources.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04
CPU: AMD Ryzen 7 5800H
GPU: Nvidia RTX 3060 Mobile
ZED SDK Version: 5.0.0
ROS: ROS2 Humble
```

### Anything else?

Here is a video showing the bug: https://youtu.be/B1RuJ6m8RGY
In this video there is console output of my app on the left (switched to zed wrapper about halfway through), and there is the rviz2 view on the right where you can see the pointcloud, depth, and rgb resetting. Additionally, you can see the mapping result of my app over top of rviz, where you can see weird duplication glitches almost every time the rviz view glitches.

Please let me know if I can provide any more context for this issue.

Myzhar, 2025-06-07:
The YouTube video is private.

alec-jensen, 2025-06-07:
@Myzhar that's my bad, just set it to unlisted

Myzhar, 2025-06-09:
The problem that you are experiencing is "frame tearing".
This can happen for different reasons:
* Bad USB 3 connectivity (long cable, bad USB 3 connector, connector screws not strenghtened, USB 3 hubs, etc)
* USB 3 buffer overflow caused by CPU overload

closed 2025-07-15 by github-actions[bot]