stereolabs/zed-ros2-wrapper #366: image topic update frequency significantly reduced in SDK 5.0

LyLewis, 2025-08-27:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

After the SDK **5.0.5** installation, I found that the topic /zed2i/zed2i/rgb/image_rect_color update frequency was **18-25hz** although the param "pub_frame_rate" was set as 60hz; whats more, the GPU was loading as 80%-90% under only the zed wrapper running.
So I rollback the SDK to **4.2.5**, with same config, same jetson orin NX 16gb, same jetpack 6.2 the topic /zed2i/zed2i/rgb/image_rect_color now can be published as **59-60hz** and GPU is loading as 40%-50%.

### Steps to Reproduce

1.launch zed wrapper
2.topic hz  /zed2i/zed2i/rgb/image_rect_color 
3.
...

### Expected Result

under 5.0.5 the frequency of topic rgb should be about 60

### Actual Result

now the frequency of topic rgb is only 20

### ZED Camera model

ZED2i

### Environment

```shell
OS:Jetson ubuntu 22.04
ROS2 HUMBLE
orin NX 16gb
```

yuhsinchan, 2025-09-06:
I got the same issue with Jetson orin nano super. It is really weird that when I use `ZED_Depth_View` depth update in 15~16 hz at 15W. However, when I launched the ROS2 wrapper, the frame rate drop to 3~4 hz. GPU usage looks the same for both case

Myzhar, 2025-09-06:
The frame rate is affected by DDS communication. 
Have you applied the fixes described in our tuning guide?
https://www.stereolabs.com/docs/ros2/dds_and_network_tuning

LyLewis, 2025-09-08:
> 日　期：2025年09月08日 18:15:10
> 主　题：Re: [stereolabs/zed-ros2-wrapper] image topic update frequency significantly reduced in SDK 5.0 (Issue #366)
>
> What's the depth mode?

neural light

阿里邮箱------------------------------------------------------------------

Myzhar, 2025-09-08:
> neural light

This is why you are not obtaining 60 FPS. The final frame rate is limited by the depth mode.
You can disable the Depth estimation if you do not need it.